simonvetter/modbus #9: How to set slaver-Id?

wwhai, 2022-05-02:
# How to set modbus slaver-Id?

simonvetter, 2022-05-02:
Hi there,

I think what you're after is the SetUnitId() method. Call it on the client whenever necessary to change the target slave id of any future requests e.g.:

```go
  client.SetUnitId(3)
  err = client.WriteRegister(0x1000, 0x20)
  regs, err = client.ReadRegister(0x1000, modbus.HOLDING_REGISTER)
  // both requests above will target slave id # 3

  // now select another slave
  client.SetUnitId(9)
  err = client.WriteRegister(0x1001, 0x30)
  err = client.WriteRegister(0x1010, 0x01)
  // those writes will target slave id # 9
```

Hope this helps,
-Simon

wwhai, 2022-05-02:
> Hi there,
>
> I think what you're after is the SetUnitId() method. Call it on the client whenever necessary to change the target slave id of any future requests e.g.:
>
> ```go
>   client.SetUnitId(3)
>   err [...]

Yes , It's my need, thanks

closed 2022-05-03 by wwhai