simonvetter/modbus #13: EOF while writing/reading registry with a delay

wajidp, 2022-07-06:
Hi
Am trying to write to read/write registry after a few seconds opening the connections but it fails with EOF

Timeout for client configuration is 30 second

if i try to read or write within 1 second it works perfectly, tried with a different client which works fine
What am i missing?

simonvetter, 2022-07-09:
Hi wajidp,

what device are you connecting to? Would that device have a configurable timeout for tcp client connections by any chance? What is your current poll interval?

My take is that the slave device times out and closes the tcp connection between two requests, which is why you're getting EOF errors.
If you're not seeing this with a third party client (which one, btw?) it is possible that this client is automatically re-connecting when disconnected.

If this theory is correct, I see 4 possible solutions:
1) configure the modbus slave with a longer idle TCP timeout (may not be easy or even possible depending on the device)
2) perform dummy reads every x seconds to keep the slave from timing out
3) handle client errors and calling Close()/Open() when getting io.EOFs
4) if your poll interval is quite long, Open() the client before each transaction and Close() it once done.

Does this help?

wajidp, 2022-07-14:
@simonvetter i was able to change the TCP timeout in the device which fixed the issue, thanks

closed 2022-07-14 by wajidp