simonvetter/modbus #18: fix: prevent addition of client connection when modbus server is not started

LucEdfsf, 2022-10-27:
On rare circumstances, the modbus server can be stop right after accepting client connection. Therefore we can add this connection to the list while the server is down

In `server.go` the current code is as follow

```
		sock, err = ms.tcpListener.Accept()
		if err != nil {
			// if the server has just been stopped, return here
			ms.lock.Lock()
			if !ms.started {
				ms.lock.Unlock()
				return
			}
			ms.lock.Unlock()
			ms.logger.Warningf("failed to accept client connection: %v", err)
			continue
		}
(-> if ms.Stop() is called right there, lock the server, everything goes bad)
		ms.lock.Lock()
		// apply a connection limit
		if uint(len(ms.tcpClients)) < ms.conf.MaxClients {
			accepted	= true
			// add the new client connection to the pool
			ms.tcpClients	= append(ms.tcpClients, sock)
		} else {
			accepted	= false
		}
		ms.lock.Unlock()
```
I guess we could do

```
		sock, err = ms.tcpListener.Accept()
		if err != nil {
			ms.logger.Warningf("failed to accept client connection: %v", err)
			continue
		}

		ms.lock.Lock()
		// apply a connection limit
		if ms.started && uint(len(ms.tcpClients)) < ms.conf.MaxClients {
			accepted	= true
			// add the new client connection to the pool
			ms.tcpClients	= append(ms.tcpClients, sock)
		} else {
			accepted	= false
		}
		ms.lock.Unlock()
```

I am not familiar in github I was not able to open the dedicated branch to fix the issue, it might be clearer in a pull review

simonvetter, 2022-11-01:
Hey Lukas, thanks for the bug report!

Nice catch. I just pushed v1.6.0 with a fix.

Mind testing it and report here? If the issue is indeed fixed, I'll let you close this.

LucEdfsf, 2022-11-03:
Thank you for the fix, will check that it work, will close when I'm sure it's ok

simonvetter, 2022-12-02:
I'm going to go ahead and close this as I believe it is fixed, but feel free to reopen if not.

Thanks again for reporting it!

closed 2022-12-02 by simonvetter