simonvetter/modbus #19: Allow starting server on existing listener

andig, 2022-10-29:
It would be nice- especially for testing- if an existing listener could be used instead of the server creating it's own listener. Doing so would also externalise the entire TLS checking/handling.

andig, 2022-11-01:
I have- for sake of speed and being opinionated in terms of which changes I needed- created a fork at https://github.com/andig/mbserver. The main changes are:

- start server on existing listener
- removed the TLS handling- this may really be wrong but I feel this could/should be part of the calling application
- removed the client- I'm typically relying on grid-x/modbus for that
- some refactoring, especially around `binary.ByteOrder`

My changes (which are really to large part removal) are in https://github.com/andig/mbserver. If this should be of any interest please let me know which PRs to start.

simonvetter, 2022-11-01:
Thanks for the heads up andig. I'll dig into your fork and see if there's anything there for me to backport.

simonvetter, 2022-11-01:
Out of interest, what custom TLS handling is your application doing?

andig, 2022-11-01:
Really no TLS handling. I've sort of come from the "I want to determine the listener" side. Once I do that I can't have the TLS handling in the library. If that breaks things it may not be the way to go but is good enough for my purposes.

andig, 2022-11-01:
I've opened https://github.com/simonvetter/modbus/pull/21 if you want to take a look. I've not rewritten the server tests yet as I'd need to migrate them to the grid-x client (or use simonvetter/modbus ;)

andig, 2022-11-01:
@simonvetter I'd be happy to open a PR for `gofmt` and maybe `golangcilint`- that would already make the diff much smaller.