simonvetter/modbus #20: Allow custom logger

andig, 2022-10-29:
Nothing to add- would be happy to provide PR.

simonvetter, 2022-11-01:
Hi andig,

the latest release (v1.6.0) allows passing a custom logger to both server and client objects, does that help?

If yes, I'll let you close this issue.

andig, 2022-11-01:
Imho the logger should be an interface, not a concrete implementation?

simonvetter, 2022-11-01:
So, I thought about it during the weekend and went back and forth over it.

I almost sided with you on it being an interface, but the golang standard lib has a fairly compact, no-frills object which everyone seems to be using, so I just went with that for the sake of clarity and simplicity.

You should be able to instantiate a bare-bones log.Logger object with the third party io.Writer of your choice though, see https://github.com/simonvetter/modbus/blob/master/logger_test.go#L13 for an example.

If that proves to be an issue over time, i'll reconsider and change the log.Logger type to an interface (which log.Logger satisfies, so as not to break the API) later on.

simonvetter, 2022-11-01:
Yep, agreed. Unfortunately Go's default logger doesn't provide any way to specify the criticity/level of a message.

That said, is your use case logging so much that logger performance is an issue? On my (medium to large) deployments, I'm only seeing the occasional CRC error or wrong transaction ID, not much more.

andig, 2022-11-01:
> That said, is your use case logging so much that logger performance is an issue?

No, it isn't. But I wouldn't use that as argument against "doing it right". I need a logger for the physical traffic and I prefer to raise PRs against libraries I'm using to introduce interfaces. Ymmv of course.

Happy to close if you want to leave as-is now.

simonvetter, 2022-11-01:
Oh yeah, i see. Like I said, I may revisit this in the future, but I'd rather keep the design simple and adhere to Golang's standard libraries for now.

If full wire traffic logging is your use case then I agree logging performance could be an issue.
If I had to implement something like this I'd most likely do it as a generic TCP proxy, or maybe as a wrapper around a TCPConn object, rather than adding it to the modbus library.

I'm going to go ahead and close this, feel free to reopen if needed.

closed 2022-11-01 by simonvetter