simonvetter/modbus #22: Modbus read velocity

odejesuscactus, 2022-12-02:
I've been trying to play with the read frequency of a Modbus sensor with the use of these library functions, and I've found myself with a limitation of 64Hz. If reading two modbus sensors, then the read frequency is limitated to 32.5Hz. Is there a way to improve this limitation? My idea is to have at least 5 sensors that can each be read at a minimum frequency of 100Hz. Thanks!

simonvetter, 2022-12-02:
Hi there,
first off, would you mind sharing more details about your setup ? Which make and model of sensors are you using ?
Are you using modbus/tcp (Ethernet) or rtu (serial) ?
If you're using serial, which serial parameters are you using ? Are those parameters (mostly baudrate) fixed, or can they be changed on the sensors ?

simonvetter, 2023-05-31:
Closing this for inactivity, feel free to reopen if needed.

closed 2023-05-31 by simonvetter