simonvetter/modbus #23: 'protocol  error' on Mac OSX

shantanoo, 2022-12-06:
Code cross-compiled on Mac OSX for windows/amd64 and linux/arm works properly. But on Mac OSX give `protocol error`.

Go version `go version go1.19.3 darwin/amd64`.

Device connected to computer (windows, arm, macosx) using same USB to serial convertor.

Same converter and device works properly with `pymodbus` python library on Mac OSX.

simonvetter, 2022-12-06:
Hi shantanoo,

mind giving me a little more info on your setup? What device are you trying to talk to? RS232 or RS485?

I've used the lib on OSX Big Sur without issue, using an USB to RS485 adapter. I've also seen people using it just fine, but that wasn't necessarily on the latest OSX releases.

Kind of a shot in the dark, but would using 2 stop bits help?

shantanoo, 2022-12-07:
Hi @simonvetter , following are additional details.

- Device is talking over RS232
- Convertor is https://www.cadyce.com/shop/cables-adapters/usb-2-0/ca-us9-usb-to-serial-converter-or-com-port/
- Stopbits 1
- Parity 0 (None)
- Databits 8
- Timeout 2 seconds
- Modbus operation read holding registers.

Let me know if there is a way to have additional logs for debugging. Can share such logs if required.

simonvetter, 2022-12-07:
Can you share the code where you're creating the client, and where you attempt to read from the device with both libraries ?
I'm almost sure it's a serial parameter issue.

shantanoo, 2022-12-08:
```
import (
	"fmt"
	"github.com/simonvetter/modbus"
	"time"
	"os"
)
```

```
	client, err = modbus.NewClient(&modbus.ClientConfiguration{
		URL:      url,
		Speed:    38400,              // default
		DataBits: 8,                  // default, optional
		Parity:   modbus.PARITY_NONE, // default, optional
		StopBits: 1,                  // default if no parity, optional
		Timeout:  5 * time.Second,
	})
```

I don't think that it is serial parameter issue, it would not have worked on linux and windows platform in that case.

shantanoo, 2022-12-08:
Yes. From the same code, i build the binary with following commands.

```
GOOS=linux GOARCH=arm go build
GOOS=windows GOARCH=amd64 go build
GOOS=darwin GOARCH=amd64 go build
```

Linux and windows binaries work, but the darwin binary fails with `protocol error`

simonvetter, 2022-12-08:
Okay. I'm going to try and find a darwin machine to test with and see if I can reproduce it.

simonvetter, 2022-12-09:
It surely could. Just wondering, if you compile natively instead of cross compiling, does it change anything ?

shantanoo, 2022-12-10:
Will check and let you know. On Mac OX i was running it with `go run .`, instead will create binary using `go build .` and check whether it works or not; though  i doubt it might behave differently :)

simonvetter, 2022-12-10:
Oh okay, I thought you only cross compiled from another OS before trying to run the binary on OSX.

If you've already tested with go run, indeed building then running won't behave any differently (when invoked with go run, the golang compiler builds just as well, the only difference being that the binary is placed in /tmp somewhere, and deleted once the program exits.)

simonvetter, 2023-07-30:
Closing this for inactivity. Feel free to reopen if needed.

closed 2023-07-30 by simonvetter