simonvetter/modbus #24: client.Close()

saimirballeshi, 2022-12-15:
Hello,
I'm opening a tcp connection and at the end of execution block I close the connection every time.
But after some opening/closing the connection I get
**dial tcp ip:port: connect: connection refused
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x5a9ab0]**

simonvetter, 2023-05-31:
Hi there,
the likely cause is that you're attempting to issue modbus requests on a client which failed to connect.

If you skip the Read/Write operations when Open() returns an error and jump right away to the next execution block, does that help ?
Now why the target device rejects the TCP connection is another issue entirely... would that device have a limit of active modbus connections ? Or is its modbus task crashing after a couple of Open()/Close() ?

You may want to re-write your code to keep the connection open between executions.

simonvetter, 2023-07-30:
Closing this for inactivity. Feel free to reopen if needed.

closed 2023-07-30 by simonvetter