simonvetter/modbus #25: uint32 LittleEndian wrong byteorder

linuxnase, 2023-02-05:
Hi Simon,
there is a bug in the byte order when calling little endian conversion binary.LittleEndian.Uint32. (encoding.go, func bytesToUint32s)
```
case LITTLE_ENDIAN:
	if wordOrder == LOW_WORD_FIRST {
		u32 = binary.LittleEndian.Uint32(in[i:i+4])
	} else {
		u32 = binary.LittleEndian.Uint32(
			[]byte{in[i+2], in[i+3], in[i+0], in[i+1]})   <--- same order as in the BIG_ENDIAN case!!
```
should be:
```
u32 = binary.LittleEndian.Uint32(
	[]byte{in[i+1], in[i+0], in[i+3], in[i+2]})
```

shuzhongling, 2023-02-20:
i think this isn't  wrong. 
in a word register, it is LITTLE endian, when read a dword, the word order is high word first.

simonvetter, 2023-05-31:
Glad you found out that everything works properly :) I'm going to go ahead and close this, feel free to reopen if needed.

closed 2023-05-31 by simonvetter