simonvetter/modbus #27: 'protocol error' on CentOS7 Ubantu20.04

wgchaoer, 2023-02-23:
modbus-cli 
[image]

The target device can normally read the value of each register by the modbus poll software of the Windows system. But only the part of the coil can be read through this modbus-cli program。

modbus_exporter: https://github.com/RichiH/modbus_exporter
telegraf.inputs: https://github.com/influxdata/telegraf/blob/release-1.24/plugins/inputs/modbus/README.md

wgchaoer, 2023-02-23:
This article mentioned the following data types about INT8
register datatype: INT8L, INT8H, UINT8L, UINT8H 
https://github.com/influxdata/telegraf/blob/release-1.25/plugins/inputs/modbus/README.md#register-datatype

and I found a similar mistake. Maybe I can find some answers from it
https://github.com/grid-x/modbus/issues/52

simonvetter, 2023-05-31:
Hi wgchaoer,

what brand and model of breaker are you seeing this with? I see you're reading input registers. If you try reading holding registers, does that help in any way? You can read holding registers with rh:<type>:<address> instead of ri:<type>:<address>.

My hunch is that the target device doesn't properly implement the modbus/tcp protocol and returns short or non-compliant data instead. This is something I've seen multiple times with RTU devices, for sure.

closed 2025-11-26 by simonvetter