simonvetter/modbus #29: Extra sleep for 3.5 character

vsaljooghi, 2023-05-14:
Hello,

Inside the function ExecuteRequest() in rtu_transport.go:

Before sending the request, master/client checks for the last activity (line 73) and if the new request violates 3.5 characters inter-frame silence, it waits till the time elapses.

`t = time.Since(rt.lastActivity.Add(rt.t35))
	if t < 0 {
		time.Sleep(t * (-1))
	}`

My question is: why master needs to wait extra at line 92, before reading/receiving the response of a newly sent request:

`	// observe inter-frame delays
	time.Sleep(rt.lastActivity.Add(rt.t35).Sub(time.Now()))`

AFAIK, every conformant modbus slave should also respect inter-frame silence. That means when we send the request, it is the responsibility of slave to respect 3.5 characters silence rather responding immediately.

Best Regards,
Vahid

simonvetter, 2023-05-31:
Hi Vahid,

you're right, a master doesn't technically need to wait for 3.5 char times before reading back a response... especially since the serial port will buffer any bytes received while we're sleeping.

This code is just an optimization to reduce CPU load on low end embedded devices: on a few embedded boards I worked with, the UART driver eats 100% of the CPU while waiting for bytes coming off the wire (someone must have implemented a busy loop), so waiting before attempting to read the response back helps a little bit.

I doubt it should impact performance in any way though : slaves should not emit anything during those 3.5 char times if they are compliant with modbus RTU.

If you think otherwise and have any data on how it affects performance on your systems, please let me know and I'll try and fix it.

vsaljooghi, 2023-06-01:
Hello,

Thanks for the response. It doesn't cause any performance issue in my case. I was just curious to know the reason for extra sleep.

MfG,
/Vahid

simonvetter, 2023-06-01:
Nice. I'm gonna go ahead and close this then, please feel free to reopen if needed. In any case thanks for reaching out!

closed 2023-06-01 by simonvetter