simonvetter/modbus #32: Write function code for HandleHoldingRegisters

shantanoo, 2023-07-17:
In TCP server implementation, how to find whether the function code was 6 or 16 (0x10)? And how to respond to it specifically?

shantanoo, 2023-07-17:
Pull Request #33

simonvetter, 2023-07-30:
Hi shantanoo,

first off, thanks for the pull request!

I'm trying to keep the code/API simple and concise, so I'm trying to understand the rationale behind this. Why would you need to differentiate between holding reg write opcodes?

shantanoo, 2023-07-31:
Need to take action based on opcode. E.g. only write multiple registers should be supported and single register should return error.

simonvetter, 2023-08-01:
> only write multiple registers should be supported and single register should return error

If what you're trying to achieve is to make sure that multiple registers are being written at once (instead of a single register), would testing on Quantity work in your handler? Something along those lines:
```go
if isWrite && req.Quantity < 2 {
  err = ErrIllegalDataValue
}
```
Note that your handler still wouldn't be able to tell if that request came through opcode 0x6 or 0x10 with a single register, but that shouldn't be a concern for your use case.

Note that this technique can also be used on reads, if you ever want to ensure that reads are atomic as well.

shantanoo, 2023-08-14:
Actually I am trying to read the modbus request on TCP server/slave and get the data from RTU client/master. In that case, server don't know what type of request comes from the client. Hence needed to differentiate between function code 6 and 16 (0x10).

simonvetter, 2024-09-22:
Wait, so you'd like to reject writes using 0x6, but not 0x10, in the TCP server, so you can reject 0x6 requests which are not supported by your RTU slave, right? Why not just use 0x10 on the RTU client then?