simonvetter/modbus #42: how to set up URL on windows OS?

crazy-cpu, 2024-03-26:
[image]

thanks!

crazy-cpu, 2024-03-26:
[image]

munnik, 2024-04-04:
I can't test this on Windows, but I think this Stack Overflow post can help: https://stackoverflow.com/questions/27909666/createfile-serial-communication-issue.
```go
client, err := modbus.NewClient(&modbus.ClientConfiguration{
    URL:  "rtu://\\\\.\\COM6",
    ...
})
```

crazy-cpu, 2024-04-08:
today.I try to fill "rtu://COM7" to URL field. it works. i can''t explain this situation.but, this problem is solved.thank you.

closed 2024-04-08 by crazy-cpu