simonvetter/modbus #44: Should be able to read 125 holding registers

jonathangjertsen, 2024-06-10:
The maximum number of registers that fits in a read is 125 while the maximum number of registers that fit in a write is 123. This package currently generates an error if you try to read 124 or 125 registers, because it uses 123 in readRegisters.  

https://github.com/simonvetter/modbus/blob/ff51403728b4e53e6fa7982f25bbf88a53854af4/client.go#L1042

pstlabdev, 2024-09-18:
I ran into this as well, however the transport of 125 was good. I updated the 123 to 125 to eliminate the error.

simonvetter, 2024-09-22:
Fixed in v1.6.3, feel free to reopen if you think otherwise.

Thanks for the report!

closed 2024-09-22 by simonvetter

pstlabdev, 2024-10-07:
Thanks for the update. For specifics from the spec on holding registers is: 123 write & 125 read. https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf