simonvetter/modbus #46: Fail on 2 digit com #'s in Windows

pstlabdev, 2024-09-18:
COM1 - 9 work as expected, however COM10, 11, etc. give an open fail. I haven't had a chance to search the code for the reason. My work around was to change COM10 to COM1 and COM11 to COM2 in Device Manager port settings/advanced for those two ports.

simonvetter, 2024-09-22:
Hi there,

thanks for the report. I'll try to see if I can reproduce this, but I'm not using windows that much, so don't hold your breath :)
In any case, the client would be passing COM10 in about the same way as it passes COM1, so I'm wondering what might be causing the issue. Maybe something with the serial library I'm using ? I'll take a look there as well.

Serhioromano, 2024-12-27:
How would you path it? I use windows but cannot get how to set a port?

```go
URL: "rtu:///COM7"
```

Like this?

simonvetter, 2024-12-27:
Try with two forward slashes instead of three? I.e. rtu://COM7

Serhioromano, 2024-12-27:
Thank you. It works fine.