simonvetter/modbus #50: Failed to get modbus data from modbus-cli on Rocky Linux 9.4

snowuyl, 2024-11-14:
I can use modpoll (https://github.com/gavinying/modpoll) to modbus data. But failed to get modbus data from modbus-cli.
Log messages of modbus-cli are as follows.
./modbus-cli --target tcp://172.16.88.110:502 rh:uint32:0x1+5 rc:0+10 wc:4:true
failed to read holding/input registers: gateway target device failed to respond
failed to read coils/discrete inputs: gateway target device failed to respond
failed to write true at coil address 0x0004: gateway target device failed to respond

Log messages of modpoll are as follows.
./modpoll -m tcp -a 2 -r 1 -c 4 -t 4:int -p none -o 1 172.16.88.110
modpoll 3.15 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2024 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP, FC3
Slave configuration...: address = 2, start reference = 1, count = 4
Communication.........: 172.16.88.110, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 32-bit integer, holding register table

-- Polling slave... (Ctrl-C to stop)
[1]: 171515394
[3]: 142679782
[5]: 7118
[7]: 0
-- Polling slave... (Ctrl-C to stop)
[1]: 171515396
[3]: 142745318
[5]: 7120
[7]: 0
-- Polling slave... (Ctrl-C to stop)
[1]: 171712003
[3]: 142876396
[5]: 7124
[7]: 0

simonvetter, 2024-11-14:
Hi there,

it looks like your device sits at unit id / slave id # 2, which you're passing to modpoll with -a 2. Since you're not selecting any unit id with modbus-cli, it defauts to 1, and there's probably no such device on the bus at this address.

Would this work any better ?
```bash
./modbus-cli --target tcp://172.16.88.110:502 sid:2 rh:uint32:0x1+5 rc:0+10 wc:4:true
```

note **sid:2** in the command above, which selects the slave id for any subsequent requests.

snowuyl, 2024-11-15:
Thanks for your reply! The following is testing result after adding sid:2.
./modbus-cli --target tcp://172.16.88.110:502 sid:2 rh:uint32:0x1+5 rc:0+10 wc:4:true
0x0001	1     : 0x09d91e39	165223993
0x0003	3     : 0x08471b66	138877798
0x0005	5     : 0x00000000	0
0x0007	7     : 0x00000000	0
0x0009	9     : 0x00000000	0
0x000b	11    : 0x00000000	0
failed to read coils/discrete inputs: illegal data address
modbus-client(172.16.88.110:502) [warn]: unexpected response code (129)
failed to write true at coil address 0x0004: protocol error

simonvetter, 2025-11-26:
I'm going to go ahead and close this since I believe you got most things figured out, feel free to reopen if needed.

closed 2025-11-26 by simonvetter