simonvetter/modbus #52: serial: let caller handle serial.ErrTimeout from serial port

ThiagoBoimer, 2024-11-28:
Hi, 

While the approach of masking `serial.ErrTimeout` might be intended to simplify higher-level logic that uses retry mechanisms, it may mask real timeout issues. Currently, I have some logic to handle reconnection and the routines were blocking because I couldn't capture this error.

**Current code:**

```go
func (spw *serialPortWrapper) Read(rxbuf []byte) (cnt int, err error) {
	// return a timeout error if the deadline has passed
	if time.Now().After(spw.deadline) {
		err = ErrRequestTimedOut
		return
	}

	cnt, err = spw.port.Read(rxbuf)
	// mask serial.ErrTimeout errors from the serial port
	if err != nil && err == serial.ErrTimeout {
		err = nil
	}

	return
}
```

I wonder if letting the user handle this wouldn't be more suitable for all use cases or if, perhaps, making this behavior configurable wouldn't be better. If you think any of this makes sense, I can submit a PR with the changes.

simonvetter, 2024-11-28:
Hi Thiago,

you're right that the design choice of hiding serial.ErrTimeout was made to simplify error handling for users.
Another goal was to avoid exposing the underlying serial library to users of the modbus client, mostly to keep the ability to swap it with another one, or maybe rewrite it, if it ever became necessary.

What are the real timeout issues you're mentioning, and how would your retry/reconnection logic handle it differently than e.g. a modbus/tcp client did not reply to a request (and thus, getting modbus.ErrRequestTimedOut)?

Did you mean that you would like your code to be able to use os.IsTimeout()?

ThiagoBoimer, 2024-11-28:
So, I did a bit more debugging and it was actually my bad. I was setting the ModbusClient timeout too high by mistake. As a result, my retry/reconnection logic never received `modbus.ErrRequestTimedOut` due to the high timeout, nor `serial.ErrTimeout` because it was masked. In the end, my routines were blocking and the logic not working.

When I removed the mask (in my fork) the retry/reconnection logic worked fine because `serial.ErrTimeout` was being returned. 

Anyway, adjusting the timeout fixed the problem and I just need to be careful not to set my logic "reconnect_interval" parameter lower than the timeout (otherwise it will hang, obviously 😅).

Thanks for the quick reply!

closed 2024-11-28 by ThiagoBoimer

simonvetter, 2024-12-05:
Glad you figured it out!