simonvetter/modbus #58: Is this project maintained?

qba73, 2025-08-17:
Hi,
Is this projects maintained? Is author willing to review and accept fixes?

I'm running into several issues, for example:

1) tests are failing:

```shell
➜  modbus-orig git:(master) go test
--- FAIL: TestTCPoverTLSClient (0.00s)
    client_tls_test.go:209: failed to load test server key pair: x509: certificate contains duplicate extension with OID "2.5.29.19"
--- FAIL: TestTLSClientOnServerTimeout (0.00s)
    client_tls_test.go:355: failed to load test server key pair: x509: certificate contains duplicate extension with OID "2.5.29.19"
modbus-server(localhost:5502) [warn]: max. number of concurrent connections reached, rejecting 127.0.0.1:52760
--- FAIL: TestTLSServer (0.00s)
    server_tls_test.go:127: failed to load test server key pair: x509: certificate contains duplicate extension with OID "2.5.29.19"
--- FAIL: TestServerExtractRole (0.00s)
    server_tls_test.go:421: failed to parse client cert: x509: certificate contains duplicate extension with OID "2.5.29.19"
tcp-transport(pipe) [warn]: received unexpected transaction id (expected 0x9218, received 0x9219)
tcp-transport(pipe) [warn]: received unexpected protocol id 0x0001
tcp-transport(pipe) [warn]: received unexpected protocol id 0x0001
FAIL
exit status 1
FAIL	github.com/simonvetter/modbus	1.392s
```

2) staticcheck issues:

```shell
➜  modbus-orig git:(master) staticcheck .
client.go:602:7: should omit comparison to bool constant, can be simplified to value (S1002)
client.go:879:2: should merge variable declaration with assignment on next line (S1021)
client_tls_test.go:236:2: this value of client is never used (SA4006)
client_tls_test.go:246:2: this value of client is never used (SA4006)
client_tls_test.go:292:2: this value of regs is never used (SA4006)
client_tls_test.go:337:2: redundant return statement (S1023)
client_tls_test.go:446:2: redundant return statement (S1023)
crc.go:46:2: redundant return statement (S1023)
crc.go:59:2: redundant return statement (S1023)
crc_test.go:67:2: redundant return statement (S1023)
crc_test.go:105:2: redundant return statement (S1023)
encoding.go:94:2: should merge variable declaration with assignment on next line (S1021)
encoding.go:166:2: should merge variable declaration with assignment on next line (S1021)
encoding_test.go:26:2: redundant return statement (S1023)
encoding_test.go:60:2: redundant return statement (S1023)
encoding_test.go:76:2: redundant return statement (S1023)
encoding_test.go:104:2: redundant return statement (S1023)
encoding_test.go:147:2: redundant return statement (S1023)
encoding_test.go:209:2: redundant return statement (S1023)
encoding_test.go:251:2: redundant return statement (S1023)
encoding_test.go:313:2: redundant return statement (S1023)
encoding_test.go:368:2: redundant return statement (S1023)
encoding_test.go:432:2: redundant return statement (S1023)
encoding_test.go:479:2: redundant return statement (S1023)
encoding_test.go:541:2: redundant return statement (S1023)
encoding_test.go:576:2: redundant return statement (S1023)
encoding_test.go:609:2: redundant return statement (S1023)
logger.go:26:2: redundant return statement (S1023)
logger.go:32:2: redundant return statement (S1023)
logger.go:38:2: redundant return statement (S1023)
logger.go:44:2: redundant return statement (S1023)
logger.go:50:2: redundant return statement (S1023)
logger.go:56:2: redundant return statement (S1023)
logger.go:63:2: redundant return statement (S1023)
logger.go:70:2: redundant return statement (S1023)
logger.go:80:2: redundant return statement (S1023)
logger_test.go:11:2: should merge variable declaration with assignment on next line (S1021)
logger_test.go:24:2: redundant return statement (S1023)
logger_test.go:29:2: should merge variable declaration with assignment on next line (S1021)
logger_test.go:42:2: redundant return statement (S1023)
modbus.go:40:2: const fcReadFileRecord is unused (U1000)
modbus.go:41:2: const fcWriteFileRecord is unused (U1000)
rtu_transport.go:92:13: should use time.Until instead of t.Sub(time.Now()) (S1024)
rtu_transport.go:255:2: redundant return statement (S1023)
rtu_transport_test.go:57:2: redundant return statement (S1023)
rtu_transport_test.go:105:2: this value of res is never used (SA4006)
rtu_transport_test.go:145:2: redundant return statement (S1023)
rtu_transport_test.go:164:2: redundant return statement (S1023)
rtu_transport_test.go:188:2: redundant return statement (S1023)
server.go:325:2: redundant return statement (S1023)
server.go:375:2: redundant return statement (S1023)
server.go:809:2: redundant return statement (S1023)
server_tcp_test.go:15:2: should merge variable declaration with assignment on next line (S1021)
server_tcp_test.go:191:2: this value of coils is never used (SA4006)
server_tcp_test.go:196:2: redundant return statement (S1023)
server_tcp_test.go:205:2: should merge variable declaration with assignment on next line (S1021)
server_tcp_test.go:352:2: this value of coils is never used (SA4006)
server_tcp_test.go:356:2: this value of coils is never used (SA4006)
server_tcp_test.go:364:2: redundant return statement (S1023)
server_tcp_test.go:439:2: this value of regs is never used (SA4006)
server_tcp_test.go:443:2: this value of regs is never used (SA4006)
server_tcp_test.go:534:2: this value of regs is never used (SA4006)
server_tcp_test.go:538:2: this value of regs is never used (SA4006)
server_tcp_test.go:556:2: this value of regs is never used (SA4006)
server_tcp_test.go:560:2: this value of regs is never used (SA4006)
server_tcp_test.go:568:2: redundant return statement (S1023)
server_tls_test.go:199:2: this value of coils is never used (SA4006)
server_tls_test.go:315:2: redundant return statement (S1023)
server_tls_test.go:571:2: redundant return statement (S1023)
tcp_transport_test.go:59:2: redundant return statement (S1023)
tcp_transport_test.go:153:2: this value of res is never used (SA4006)
tcp_transport_test.go:201:2: this value of res is never used (SA4006)
tcp_transport_test.go:209:2: redundant return statement (S1023)
tcp_transport_test.go:281:9: possible nil pointer dereference (SA5011)
	tcp_transport_test.go:278:5: this check suggests that the pointer can be nil
tcp_transport_test.go:305:2: redundant return statement (S1023)
tcp_transport_test.go:314:26: should use make(chan bool) instead (S1019)
tcp_transport_test.go:319:3: should merge variable declaration with assignment on next line (S1021)
tcp_transport_test.go:347:3: redundant return statement (S1023)
tcp_transport_test.go:372:2: redundant return statement (S1023)
tls_utils_test.go:74:2: this value of cp is never used (SA4006)
tls_utils_test.go:94:2: this value of cp is never used (SA4006)
tls_utils_test.go:117:2: this value of cp is never used (SA4006)
tls_utils_test.go:150:2: redundant return statement (S1023)
udp_test.go:165:2: redundant return statement (S1023)
```

Before I invest time I want to know if fixes to the above will be accepted.

sansmoraxz, 2025-11-06:
Looks like @simonvetter is probly not using github anymore.

I have been looking for a good modbus library in go for a while now. This was more promising than the @goburrow one (which has also gone dark), and has even more bugs.

Would anyone be interesting in maintaining a fork?

qba73, 2025-11-06:
Thanks for clarifying @simonvetter

closed 2025-11-26 by simonvetter